ros-drivers/velodyne #499: What is the focal distance unit of HDL 64e? meter or centimeter?

HappengorC, 2022-12-27:
Hi,

I want to ask what is the focal distance unit of HDL 64e in the HDL 64e XML?  https://github.com/ros-drivers/velodyne/blob/16be7978ee0fa78818a9ff04faf5efac83ccf378/velodyne_pointcloud/params/64e_s3-xiesc.yaml#L6 

Question:
**Is the unit of focal distance meter?** 

I assumed the focal distance has the unit of meter after I read the source code. 

The following equation can compensate for the intensity according to the HDL 64e s3 user manual:

https://github.com/ros-drivers/velodyne/blob/16be7978ee0fa78818a9ff04faf5efac83ccf378/velodyne_pointcloud/src/lib/rawdata.cc#L488-L493

But in this case, the focal distance is assumed to have the unit of centimeter and _corrections.focal_distance_ has the unit of meter. Hence, I think it would be 
` float focal_offset = 256  
                    * (1 - corrections.focal_distance / 131)  
                    * (1 - corrections.focal_distance / 131); `
, if corrections.focal_distance has the unit of meter.

from-Pb2Ag, 2023-05-18:
Hi, I am working with [nuScenes](https://www.nuscenes.org/nuscenes) dataset which uses a HDL-32E lidar. When considering the intensity calibration, I find it is interesting when compared to the aforementioned HDL-64E lidar: which has **channel-specific** `focal_distance`, and compute `focal_offset` with some equation and agrees with the [manual](https://yiqi-oss.oss-cn-hangzhou.aliyuncs.com/aliyun/900102614/technical_file/file_206099.pdf).

However, it seems `focal_distance` is **channel-nonspecific** for HDL-32E as the following code shows, which is also not addressed in corresponding manual. And I wonder whether it has some method to **compensate for the intensity** just like what we have done with HDL-64E, or it is infeasible (or unnecessary.) to do so. Could you offer me some insight? Thanks a lot.
https://github.com/ros-drivers/velodyne/blob/4dc95acbdfa31bacc1844be8d6a4f92517706ca9/velodyne_pointcloud/params/32db.yaml#L1-L5

JWhitleyWork, 2023-05-27:
I'm sorry but I don't know enough technical details about the hardware implementation to be able to answer this. It would be helpful if someone could reach out to Velodyne to ask this question and post a response here.