ros-drivers/velodyne #500: Option to build a 'generic' version of velodyne_rawdata (ROS free)

JChatillon, 2023-01-17:
Velodyne does not provide an 'official' HPF (HDL Packet Format) parser in c/c++, and I wonder if this could be it.

I understand that this is a ROS Driver, so it make absolute sense to have ROS interconnected, but for example, I do not always need to read ROS in my applications (ex: post-processing of pcap data), and so being "forced" to use ROS (and so Linux) to parse the data packet is not ideal. Sure I can write my own HPF decoder but it would be reinventing the wheel.

So if possible it would be nice separate the basic packet decoder (RawData::unpack()) from the ros message and to be able to build (with CMake) as target without ROS dependency. Even better if we can avoid boost dependency too :D

Maybe conditional compilation but it introduces "ugly" code like:
```c++
#ifndef VELODYNE_RAWDATA_GENERIC
#include <ros/ros.h>
#include <ros/package.h>
#endif
...
#ifndef VELODYNE_RAWDATA_GENERIC
ROS_DEBUG_STREAM("Received packet, time: " << pkt.stamp);
#endif
````
so maybe an OOP would help by using an interface.

pedroexenberger, 2023-02-02:
Not the solution you want perhaps, but I managed to use ROS packages as c++ libraries without having to run roscore with the help of this repository: https://github.com/gerkey/ros1_external_use. You still need to install the ROS packages (to get the libraries installed), but no need to have ROS running at runtime, just use its libraries!

JWhitleyWork, 2023-02-02:
I have no problem with this repository being reorganized to seperate the parsing algorithms from the ROS-specific code. However, this is a massive undertaking at current and I simply don't know of anyone who shares this idea and also has the resources to actually implement it. If you are such a person, please open a new issue as a feature request with a plan for how to do it and I'm happy to review.

closed 2023-02-02 by JWhitleyWork