ros-drivers/velodyne #505: Keep pull timeout() while wireshark can receive the data and lidar can be ping

TZECHIN6, 2023-03-29:
**Please complete the following information:**
 - OS and Version: Ubuntu 22.04
 - ROS Version: Humble
 - Built from Source or Downloaded from Official Repository: Build from this repo -b ros2
 - Version: 00d793b5d8c760bd616c32789ead317c544807e5

**Describe the bug**
Not sure is it a driver bug or what... I have bought so many velodyne lidar (at least got 8 lidars of VLP-16). So I do successfully setup the lidar in before.

However, for the one I encountered the issue was that I can ping and wireshark capture the package without issue. The driver installation for ROS2 also looks fine, however it keeps pull timeout(), I have also tried to use veloview to see it still no lidar scan output.

**To Reproduce**
Steps to reproduce the behavior:
1. Install as usual
2. git clone the repo -b ros2 into own workspace
3. rosdep install
4. colcon build
5. source it
6. ros2 launch ...

**Expected behavior**
Expected to receive lidar scan.

**Additional context**
Here's the config, as I know most of you would think this is a ip or port issue... I also think so but seem not.
LiDAR (VLP16):
- IP: 192.168.1.201
- port 2368 and 8308
- Destination IP: 192.168.1.100/24

PC:
- IP: 192.168.1.100/24
- Gateway: 192.168.1.1
- No firewall is enabled

Ping works fine and I can see the UDP package from wireshark (length=1206)

Any idea I can debug this issue? as this is the first time I encounter this, I solved this similar issue in before by setting the correct ip and port, but not this time....

TZECHIN6, 2023-03-29:
Solved: make sure your power adapter is 12V 3A. I got a 2A and its not working

TZECHIN6, 2023-04-13:
I need help on this issue, the situation is very unstable.

I can no longer launch the velodyne driver as it keeps `[velodyne_driver_node-1] [WARN] [1681360614.646165429] [velodyne_driver_node]: Velodyne poll() timeout`. I have checked the IP setting in both PC and VLP16 LiDAR, it can be pinged.

Anyone has solution how to debug this issue? As everything seem make sense to me, no clues to debug this timeout issue.

TZECHIN6, 2023-04-14:
Not sure the root cause, I somehow get the lidar point back with all the same setting and codes.

1. Head to the LiDAR web interface
2. Set destination IP as 255.255.255.255 (previous is set to 192.168.1.100 which is my PC's IP)
3. Click "set" and "save config"
4. Now LiDAR can be seen
5. If you want to use back destination IP
6. Head back to the web interface and set the destination IP to 192.168.1.100 and click set and save config
7. Now LiDAR can be seen

To be honest... this is quite silly.

closed 2023-04-14 by TZECHIN6

DipFlip, 2024-09-19:
For anyone coming here with the `poll timout()` response, what worked for me on Ubuntu was 
```bash
sudo ufw allow from 192.168.1.201
```