ros-drivers/velodyne #516: How to modify the max RPM to 1200.

abhishekt711, 2023-05-29:
In the velodyne VLS128 web interface maximum RPM is limited to 600. But velodyne VLS128 claims to work at 1200RPM too. How can I modify the max RPM value to 1200 ? 

[image]

JWhitleyWork, 2023-05-29:
I have never worked with an Alpha Prime and I don't have the user manual. If you change the RPM field to 1200 and click SET, does anything happen?

abhishekt711, 2023-05-29:
It automatically keep 600. I think if I do firmware update then it will update. But I couldn’t find firmware update .flash file for VLS128 on their website

JWhitleyWork, 2023-06-01:
@abhishekt711 You need to contact Velodyne directly to get the firmware update file.