ros-drivers/velodyne #522: Didn't work on VLP16

jaeseok4104, 2023-06-06:
**Please complete the following information:**
 - OS and Version: ubuntu18.04
 - ROS Version: melodic
 - Built from Source or Downloaded from Official Repository:
 - Version: [if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-velodyne`, if from source, give commit hash]

**Describe the bug**
[image]
didn't published sensor_msgs/pointcloud2
how can i fix this error?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

JWhitleyWork, 2023-06-07:
For your target frame to be `odom`, there must be a valid set of transforms between the velodyne's frame (by default, `velodyne`) and the `odom` frame. Is this true for your setup?

jaeseok4104, 2023-06-07:
how can i set odom to velodyne frame? when i use this driver, should i make odom to velodyne tf?

jaeseok4104, 2023-06-07:
I didn't understand what you said. I think this driver posts point cloud in velodyne frames, and I think if I set to velodyne frames in rviz, I should be able to see point cloud data.

JWhitleyWork, 2023-06-16:
> how can i set odom to velodyne frame? when i use this driver, should i make odom to velodyne tf?

If you want to use an `/odom` frame then yes, you need to publish a TF between `/velodyne` (the default frame for the sensor) and `/odom`. If you do not want to use `/odom` then you can just change `rviz` to view in `/velodyne` frame.

JWhitleyWork, 2023-06-16:
> when i use ros-noetic i could used this driver. don't surrport melodic version?

`melodic` was supported until 2 weeks ago when `melodic` became end-of-life. The driver was confirmed to work on `melodic` by many users for a very long time. If you still want to use `melodic`, you can use the `melodic-devel` branch of this repository or the `apt` package.