ros-drivers/velodyne #523: Non-eth0 Interface

guidout, 2023-06-10:
Hello,
I have set the ip on my eth card and I can ping the VLP-16 address (192.168.1.201), I can also access the interface from the web.
But the ros driver cannot connect to it. Below is my output.
My interface name is enx8cae4cd75ae1 but I read from another Issue here that the name shouldn't matter...not sure what to do.

```
$ ros2 launch velodyne_driver velodyne_driver_node-VLP16-launch.py device_ip:=192.168.1.201
[INFO] [launch]: All log files can be found below /home/gritelli/.ros/log/2023-06-10-07-41-11-122924-techdev1-8623
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [velodyne_driver_node-1]: process started with pid [8625]
[velodyne_driver_node-1] [INFO] [1686397271.174903132] [velodyne_driver_node]: Velodyne VLP-16 rotating at 600.000000 RPM
[velodyne_driver_node-1] [INFO] [1686397271.174944562] [velodyne_driver_node]: publishing 76 packets per scan
[velodyne_driver_node-1] [INFO] [1686397271.174949938] [velodyne_driver_node]: Cut at specific angle feature activated. Cutting velodyne points always at 6.283185 rad.
[velodyne_driver_node-1] [INFO] [1686397271.174952708] [velodyne_driver_node]: expected frequency: 9.921 (Hz)
[velodyne_driver_node-1] [INFO] [1686397271.174978626] [velodyne_driver_node]: Only accepting packets from IP address: 192.168.1.201
[velodyne_driver_node-1] [INFO] [1686397271.174982772] [velodyne_driver_node]: Opening UDP socket: port 2368
[velodyne_driver_node-1] [WARN] [1686397272.176334285] [velodyne_driver_node]: Velodyne poll() timeout
[velodyne_driver_node-1] [WARN] [1686397273.177630591] [velodyne_driver_node]: Velodyne poll() timeout
[velodyne_driver_node-1] [WARN] [1686397274.178103832] [velodyne_driver_node]: Velodyne poll() timeout
```

guidout, 2023-06-10:
Problem solved. In Ubuntu I set my computer IP to 192.168.1.100 but, from the Velodyne web interface, I noticed the host IP was 192.168.1.3. Setting the IP to the .3 made it work. I'm not sure why they are different. 
Please close this issue and sorry for the false alarm. Hopefully, it will be useful for somebody else though.

SeifShams, 2023-07-22:
how do you solve it? because I have the same problem