ros-drivers/velodyne #532: I can't run two lidar at the same time

leeyuelun, 2023-09-27:
**Please complete the following information:**
 - OS and Version: [e.g. Ubuntu 18.04]
 - ROS Version: [melodic]
 - Built from Source or Downloaded from Official Repository:
 -
I can run run lidar each other, but can't at the same time.
it awlays output 
[ WARN] [1695786609.867868744]: Velodyne poll() timeout
[ERROR] [1695786609.867924776]: DriverNodelet::devicePoll - Failed to poll device.

what's problem ?
this is my rosnode list
/rosout
/velodyne_1/velodyne_1_nodelet_manager
/velodyne_1/velodyne_1_nodelet_manager_driver
/velodyne_1/velodyne_1_nodelet_manager_laserscan
/velodyne_1/velodyne_1_nodelet_manager_transform
/velodyne_2/velodyne_2_nodelet_manager
/velodyne_2/velodyne_2_nodelet_manager_driver
/velodyne_2/velodyne_2_nodelet_manager_laserscan
/velodyne_2/velodyne_2_nodelet_manager_transform

this my launch file
`

<launch>

  
  <arg name="64e_1_calibration" default="$(find velodyne_pointcloud)/params/64e_s3-xiesc.yaml"/>
  <arg name="64e_1_device_ip" default="172.20.102.31" />
  <arg name="64e_1_frame_id" default="velodyne_1" />
  <arg name="64e_1_manager" default="$(arg 64e_1_frame_id)_nodelet_manager" />
  <arg name="64e_1_port" default="2368" />
  <arg name="64e_2_calibration" default="$(find velodyne_pointcloud)/params/64e_s3-xiesc.yaml"/>
  <arg name="64e_2_device_ip" default="172.20.2.32" />
  <arg name="64e_2_frame_id" default="velodyne_2" />
  <arg name="64e_2_manager" default="$(arg 64e_2_frame_id)_nodelet_manager" />
  <arg name="64e_2_port" default="2368" />

  <arg name="max_range" default="130.0" />
  <arg name="min_range" default="0.4" />
  <arg name="pcap" default="" />
  <arg name="read_fast" default="false" />
  <arg name="read_once" default="false" />
  <arg name="repeat_delay" default="0.0" />
  <arg name="rpm" default="600.0" />
  <arg name="gps_time" default="false" />
  <arg name="pcap_time" default="false" />
  <arg name="cut_angle" default="-0.01" />
  <arg name="timestamp_first_packet" default="false" />
  <arg name="laserscan_ring" default="-1" />
  <arg name="laserscan_resolution" default="0.007" />
  <arg name="model" value="64E_S3"/>

  
 <group ns = "$(arg 64e_1_frame_id)">
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="device_ip" value="$(arg 64e_1_device_ip)"/>
    <arg name="frame_id" value="$(arg 64e_1_frame_id)"/>
    <arg name="manager" value="$(arg 64e_1_manager)" />
    <arg name="port" value="$(arg 64e_1_port)"/>
    <arg name="model" value="$(arg model)"/>
    <arg name="pcap" value="$(arg pcap)"/>
    <arg name="read_fast" value="$(arg read_fast)"/>
    <arg name="read_once" value="$(arg read_once)"/>
    <arg name="repeat_delay" value="$(arg repeat_delay)"/>
    <arg name="rpm" value="$(arg rpm)"/>
    <arg name="gps_time" value="$(arg gps_time)"/>
    <arg name="pcap_time" value="$(arg pcap_time)"/>
    <arg name="cut_angle" value="$(arg cut_angle)"/>
    <arg name="timestamp_first_packet" value="$(arg timestamp_first_packet)"/>
  </include>

  
  <include file="$(find velodyne_pointcloud)/launch/transform_nodelet.launch">
    <arg name="model" value="$(arg model)"/>
    <arg name="calibration" value="$(arg 64e_1_calibration)"/>
    <arg name="manager" value="$(arg 64e_1_manager)" />
    <arg name="fixed_frame" value="" />
    <arg name="target_frame" value="" />
    <arg name="max_range" value="$(arg max_range)"/>
    <arg name="min_range" value="$(arg min_range)"/>
  </include>

  
  <include file="$(find velodyne_pointcloud)/launch/laserscan_nodelet.launch">
    <arg name="manager" value="$(arg 64e_1_manager)" />
    <arg name="ring" value="$(arg laserscan_ring)"/>
    <arg name="resolution" value="$(arg laserscan_resolution)"/>
  </include>
 </group>

 <group ns = "$(arg 64e_2_frame_id)">
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="device_ip" value="$(arg 64e_2_device_ip)"/>
    <arg name="frame_id" value="$(arg 64e_2_frame_id)"/>
    <arg name="manager" value="$(arg 64e_2_manager)" />
    <arg name="port" value="$(arg 64e_2_port)"/>
    <arg name="model" value="$(arg model)"/>
    <arg name="pcap" value="$(arg pcap)"/>
    <arg name="read_fast" value="$(arg read_fast)"/>
    <arg name="read_once" value="$(arg read_once)"/>
    <arg name="repeat_delay" value="$(arg repeat_delay)"/>
    <arg name="rpm" value="$(arg rpm)"/>
    <arg name="gps_time" value="$(arg gps_time)"/>
    <arg name="pcap_time" value="$(arg pcap_time)"/>
    <arg name="cut_angle" value="$(arg cut_angle)"/>
    <arg name="timestamp_first_packet" value="$(arg timestamp_first_packet)"/>
  </include>

  
  <include file="$(find velodyne_pointcloud)/launch/transform_nodelet.launch">
    <arg name="model" value="$(arg model)"/>
    <arg name="calibration" value="$(arg 64e_2_calibration)"/>
    <arg name="manager" value="$(arg 64e_2_manager)" />
    <arg name="fixed_frame" value="" />
    <arg name="target_frame" value="" />
    <arg name="max_range" value="$(arg max_range)"/>
    <arg name="min_range" value="$(arg min_range)"/>
  </include>

  
  <include file="$(find velodyne_pointcloud)/launch/laserscan_nodelet.launch">
    <arg name="manager" value="$(arg 64e_2_manager)" />
    <arg name="ring" value="$(arg laserscan_ring)"/>
    <arg name="resolution" value="$(arg laserscan_resolution)"/>
  </include>
 </group>

</launch>`

JWhitleyWork, 2023-09-27:
If both lidar are on the same network, you must use the web interface to configure different IP address and ports for each lidar.

leeyuelun, 2023-09-27:
> If both lidar are on the same network, you must use the web interface to configure different IP address and ports for each lidar.

I got different IP 
lidar1 IP:172.20.102.31 port:2368
lidar2 IP:172.20.2.32 port:2368

my lidar is hdl64e so it can't use web interface and only can change IP not for port

JWhitleyWork, 2023-09-27:
Even if the devices have different IP addresses, if they are both broadcasting to the same port, you will receive the data from both lidars on that one port. You must either separate them into different physical networks or change the port. On HDL-64Es, I believe this must be done using the serial (RS232) interface.

However, I don't believe this is the cause of your problem. Try running two separate launch files instead of one launch file and see if you get the same error.