ros-drivers/velodyne #534: Release v2.4.0 to Iron

doisyg, 2023-10-15:
**Is your feature request related to a problem? Please describe.**
We had some issue on our Humble -> Iron transition due to inconsistent released binaries version.
Current Humble is v2.4.0 but Iron is v2.3.0

**Describe the solution you'd like**
Release v2.4.0 to Iron

**Describe alternatives you've considered**
Building from source

doisyg, 2023-11-13:
Hello @JWhitleyWork, 
Any blocking points to trigger a v2.4.0 release to Iron ?
I was hoping to see it appears in the last Iron sync. It would be nice if we did not have to build it from source :pray: . Let us now if we can help (cc @nachovizzo)

nachovizzo, 2023-11-13:
> Hello @JWhitleyWork, Any blocking points to trigger a v2.4.0 release to Iron ? I was hoping to see it appears in the last Iron sync. It would be nice if we did not have to build it from source 🙏 . Let us now if we can help (cc @nachovizzo)

Also adding @mikepurvis and @wjwwood to the conversation :)

ruffsl, 2024-05-28:
V2.4.0 now seems to be listed as the current version for non-EOL ROS 2 distros, including Iron, Jazzy, and Rolling:
- https://index.ros.org/p/velodyne_driver/#iron

Although, I don't know where it is now being released from, as this repo seem inactive:
- https://github.com/ros-drivers-gbp/velodyne-release