ros-drivers/velodyne #541: velodyne-128 driver issue

YangJunghoon, 2024-09-23:
**Please complete the following information:**
 - OS and Version: [Ubuntu 22.04]
 - ROS Version: [humble]
 - Built from Source or Downloaded from Official Repository:
 - Version: sudo apt install ros-humble-velodyne ros-humble-pcl-ros ros-humble-visualization-msgs

**Describe the bug**
velodyne_transform_node-2] [INFO] [1727069870.757835119] [velodyne_transform]: correction angles: config/VLS128.yaml
[velodyne_driver_node-1] terminate called after throwing an instance of 'std::runtime_error'
[velodyne_driver_node-1]   what():  Unknown Velodyne LIDAR model: VLS-128

**To Reproduce**
Steps to reproduce the behavior:
1. I found out that you guys updated the vls-128 version.
2. so i setup through sudo apt install ros-humble-velodyne ros-humble-pcl-ros ros-humble-visualization-msgs 
3. and make nodes like, ex) Node(
            package='velodyne_driver',
            executable='velodyne_driver_node',
            name='velodyne_driver',
            parameters=[{
                'frame_id': 'velodyne',
                'model': 'VLS-128'
            }],
            output='screen'
        ),
        Node(
            package='velodyne_pointcloud',
            executable='velodyne_transform_node',
            name='velodyne_transform',
            parameters=[{
                'min_range': 0.4,
                'max_range': 130.0,
                'calibration': 'config/VLS128.yaml',
                'model': 'VLS-128'
            }],
            output='screen'
        )
4. but i got [velodyne_driver_node-1] terminate called after throwing an instance of 'std::runtime_error'
[velodyne_driver_node-1]   what():  Unknown Velodyne LIDAR model: VLS-128
[velodyne_transform_node-2] [WARN] [1727069870.763872376] [velodyne_pointcloud]: Timings not supported for model VLS-128
[velodyne_transform_node-2] [WARN] [1727069870.763890396] [velodyne_pointcloud]: NO TIMING OFFSETS CALCULATED. ARE YOU USING A SUPPORTED VELODYNE SENSOR?
[ERROR] [velodyne_driver_node-1]: process has died [pid 90514, exit code -6, cmd '/opt/ros/humble/lib/velodyne_driver/velodyne_driver_node --ros-args -r __node:=velodyne_driver --params-file /tmp/launch_params_lzog4cqa'].
5. i checked up ros/humble/share/velodyne_driver/launch, then  i found velodyne_driver_node-VLP16-composed-launch.py,velodyne_driver_node-VLP16-launch.py, velodyne_driver_node-VLP32C-composed-launch.py, velodyne_driver_node-VLP32C-launch.py but o couldn't found 128, 
6. How do I use version 128

**Expected behavior**
using VLS-128 driver

valgur, 2024-09-23:
Set the model ID to `VLS128` to fix the issue.

YangJunghoon, 2024-09-23:
i tryed that
Node(
            package='velodyne_driver',
            executable='velodyne_driver_node',
            name='velodyne_driver',
            parameters=[{
                'frame_id': 'velodyne',
                'model': 'VLS128'
            }],
            output='screen'
        ),
        Node(
            package='velodyne_pointcloud',
            executable='velodyne_transform_node',
            name='velodyne_transform',
            parameters=[{
                'min_range': 0.4,
                'max_range': 130.0,
                'calibration': 'config/VLS128.yaml',
                'model': 'VLS128'
            }],
            output='screen'
        )
but return error
[velodyne_transform_node-2] [INFO] [1727073236.440597056] [velodyne_transform]: correction angles: config/VLS128.yaml
[velodyne_driver_node-1] terminate called after throwing an instance of 'std::runtime_error'
[velodyne_driver_node-1]   what():  Unknown Velodyne LIDAR model: VLS128
[velodyne_transform_node-2] [WARN] [1727073236.447534543] [velodyne_pointcloud]: Timings not supported for model VLS128
[velodyne_transform_node-2] [WARN] [1727073236.447553540] [velodyne_pointcloud]: NO TIMING OFFSETS CALCULATED. ARE YOU USING A SUPPORTED VELODYNE SENSOR?
[ERROR] [velodyne_driver_node-1]: process has died [pid 94803, exit code -6, cmd '/opt/ros/humble/lib/velodyne_driver/velodyne_driver_node --ros-args -r __node:=velodyne_driver --params-file /tmp/launch_params_t0ly_cme'].

YangJunghoon, 2024-09-23:
i tried that
Node(
package='velodyne_driver',
executable='velodyne_driver_node',
name='velodyne_driver',
parameters=[{
'frame_id': 'velodyne',
'model': 'VLS128'
}],
output='screen'
),
Node(
package='velodyne_pointcloud',
executable='velodyne_transform_node',
name='velodyne_transform',
parameters=[{
'min_range': 0.4,
'max_range': 130.0,
'calibration': 'config/VLS128.yaml',
'model': 'VLS128'
}],
output='screen'
)
but return error
[velodyne_transform_node-2] [INFO] [1727073236.440597056] [velodyne_transform]: correction angles: config/VLS128.yaml
[velodyne_driver_node-1] terminate called after throwing an instance of 'std::runtime_error'
[velodyne_driver_node-1] what(): Unknown Velodyne LIDAR model: VLS128
[velodyne_transform_node-2] [WARN] [1727073236.447534543] [velodyne_pointcloud]: Timings not supported for model VLS128
[velodyne_transform_node-2] [WARN] [1727073236.447553540] [velodyne_pointcloud]: NO TIMING OFFSETS CALCULATED. ARE YOU USING A SUPPORTED VELODYNE SENSOR?
[ERROR] [velodyne_driver_node-1]: process has died [pid 94803, exit code -6, cmd '/opt/ros/humble/lib/velodyne_driver/velodyne_driver_node --ros-args -r __node:=velodyne_driver --params-file /tmp/launch_params_t0ly_cme'].

valgur, 2024-09-23:
Ok, that's a bug in the driver then: https://github.com/ros-drivers/velodyne/blob/d8cf623a922b1f12995e8c71295924c2905bd9a3/velodyne_pointcloud/src/lib/rawdata.cpp#L188-L210

Btw, you can give https://github.com/valgur/velodyne_decoder a try. It's not a direct replacement, but the precise timing info for VLS-128 is definitely supported there, together with dual-return support and other improvements.

rezagolestani, 2025-09-20:
One question out of topic, what set up you use to power up the VLS-128? Do you use the interface box?