ros-drivers/velodyne #550: Eigen3 and PCL are not being exported properly by velodyne_pointcloud

benjaminholdennearearth, 2024-11-05:
**Please complete the following information:**
 - OS and Version: Ubuntu 22.04
 - ROS Version: Humble
 - Built from Source or Downloaded from Official Repository: Official Repository
 - Version: 2.5.1-1jammy.20241031.171005

**Describe the bug**
Eigen and PCL are not being exported properly by velodyne_pointcloud
* [CMakeLists.txt Line 154](https://github.com/ros-drivers/velodyne/blob/ros2/velodyne_pointcloud/CMakeLists.txt#L154) should be "Eigen3"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [velodyne meta package CMakeLists.txt](https://github.com/ros-drivers/velodyne/blob/ros2/velodyne/CMakeLists.txt)
2. Add a line: `find_package(velodyne_pointcloud REQUIRED)`
3. Try to build `colcon build --packages-up-to velodyne`
4. See error about "By not providing "Findeigen.cmake" in CMAKE_MODULE_PATH this project..."
5. Change [velodyne_pointcloud CMakeLists.txt line: 154](https://github.com/ros-drivers/velodyne/blob/ros2/velodyne_pointcloud/CMakeLists.txt#L154) to `Eigen3`
6. Remove previous build artifacts
7. Rebuild (step 3)
8. See that Eigen is now fine but the same error appears only about PCL this time
9. Change [velodyne_pointcloud CMakeLists.txt line: 157](https://github.com/ros-drivers/velodyne/blob/ros2/velodyne_pointcloud/CMakeLists.txt#L157) to `PCL`
10. Remove previous build artifacts
11. Rebuild (step 3)
12. View that all errors are gone now

**Expected behavior**
Eigen3 and PCL should be exported properly by ament so that any package that has a build dependency on "velodyne_pointcloud" can build properly

**Additional context**
I believe that in general the name of the package used in the `find_package(...)` command should match the name (and case) used in the `ament_export_dependencies` command. 

I hope this issue request is helpful and thank you for the hard work you all put into this software.

hect95, 2024-11-06:
Hi,

I am also experiencing this in Ubuntu 22.04 😢

benjaminholdennearearth, 2024-11-06:
Hi Hector! While this issue is open here is how you can work around this.
1. Comment out your `find_package(velodyne_pointcloud)` call in your `CMakeLists.txt` file
2. Add the following code to your `CMakeLists.txt` file to "manually" find the package
```cmake
# Manually pull in velodyne_pointcloud libraries "velodyne_rawdata" and "velodyne_cloud_types"
# - Change the headers in `find_path(...)` if other headers are needed
# - Change the libraries in `find_library(...)` if other libraries are needed
find_path(
  LIBVELODYNE_POINTCLOUD_INCLUDE_DIR NAMES
    calibration.hpp
    convert.hpp
    datacontainerbase.hpp
    organized_cloudXYZIRT.hpp
    pointcloudXYZIRT.hpp
    rawdata.hpp
    transform.hpp
  PATH_SUFFIXES velodyne_pointcloud)
find_library(LIBVELODYNE_POINTCLOUD_LIBRARY NAMES velodyne_rawdata velodyne_cloud_types)
include(FindPackageHandleStandardArgs)
find_package_handle_standard_args(libvelodyne_pointcloud DEFAULT_MSG LIBVELODYNE_POINTCLOUD_LIBRARY LIBVELODYNE_POINTCLOUD_INCLUDE_DIR)
mark_as_advanced(LIBVELODYNE_POINTCLOUD_LIBRARY LIBVELODYNE_POINTCLOUD_INCLUDE_DIR)
add_library(libvelodyne_pointcloud::libvelodyne_pointcloud INTERFACE IMPORTED GLOBAL)
target_link_libraries(libvelodyne_pointcloud::libvelodyne_pointcloud INTERFACE ${LIBVELODYNE_POINTCLOUD_LIBRARY})
target_include_directories(libvelodyne_pointcloud::libvelodyne_pointcloud INTERFACE ${LIBVELODYNE_POINTCLOUD_INCLUDE_DIR})
```

Then later when creating your library do something like this
```cmake
add_library(my_velodyne_library src/my_velodyne_library.cpp)
ament_target_dependencies(my_velodyne_library ${my_dependencies}) # do not include velodyne_pointcloud here
target_include_directories(my_velodyne_library
  PUBLIC
  $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>
  $<INSTALL_INTERFACE:include/${PROJECT_NAME}>
)
target_link_libraries(my_velodyne_library libvelodyne_pointcloud::libvelodyne_pointcloud) # now link to velodyne_pointcloud
```

hect95, 2024-11-10:
> Hi Hector! While this issue is open here is how you can work around this.
>
> 1. Comment out your `find_package(velodyne_pointcloud)` call in your `CMakeLists.txt` file
> 2. Add the following code to [...]

Hi @benjaminholdennearearth 

Thanks for sharing this!

As I'm running this driver inside a Docker container using the Debian version of the `velodyne_pointcloud` package, and based on your initial suggestions on how to resolve this in the source code, I was able to build my launch package, which depends on velodyne_pointcloud, with the following:

```
sed -i 's/\beigen\b/Eigen3/g' /opt/ros/humble/share/velodyne_pointcloud/cmake/ament_cmake_export_dependencies-extras.cmake 

sed -i 's/\bpcl\b/PCL/g' /opt/ros/humble/share/velodyne_pointcloud/cmake/ament_cmake_export_dependencies-extras.cmake
```

I don't think this is the ideal approach, but it should keep me going until this issue is resolved  😅